BubTec/DPVControl #10: [BUG]  Branch smooth_acceleration

n3roGit, 2023-11-14:
Change SetDuty like this:

Start motor speed not 0. make it 0.2
Min speed: 0.3
max speed 1.0

happyherp, 2023-11-15:
Try version cb04071f92cc9727334097aa8c9bbf7584ddee37

The motor will now start right at 0.2 and then accelerate slowly as usual. Slowdown is unchanged. 

```

14:09:21.709 -> targetMotorSpeed: 0.37
14:09:21.709 -> currentMotorSpeed: 0.20
14:09:21.740 -> currentMotorSpeed: 0.21
14:09:21.834 -> currentMotorSpeed: 0.22
14:09:21.880 -> currentMotorSpeed: 0.23
14:09:21.927 -> currentMotorSpeed: 0.24
14:09:21.974 -> currentMotorSpeed: 0.25
```

n3roGit, 2023-11-15:
I have adjusted MIN_DUTY_SOFT slightly so that it works well. But now we need a delay before the motor is started, because otherwise it wants to start directly with every click or double-click.

n3roGit, 2023-11-15:
i have just noticed something else. if i set MIN_SPEED_DUTY to 0.38, for example, which corresponds to 6000 erpm, then it starts at gear 1 (lowest level) at a value of approx. 0.44.

speed down: 1
targetMotorSpeed: 0.44
currentMotorSpeed: 0.49
currentMotorSpeed: 0.48

happyherp, 2023-11-15:
> I have adjusted MIN_DUTY_SOFT slightly so that it works well. But now we need a delay before the motor is started, because otherwise it wants to start directly with every click or double-click.

cdc9fff52e625ef821b267f744e2a8ba9262cd31 adds a delay between holding down a lever and the motor switching from off to on.

n3roGit, 2023-11-15:
> > I have adjusted MIN_DUTY_SOFT slightly so that it works well. But now we need a delay before the motor is started, because otherwise it wants to start directly with every click or double-click.
> 
> [cdc9fff](https://github.com/n3roGit/DPVControl/commit/cdc9fff52e625ef821b267f744e2a8ba9262cd31) adds a delay between holding down a lever and the motor switching from off to on.

delay works great!

closed 2023-11-16 by happyherp