BubTec/DPVControl #11: [BUG]  Stepwise Up/Down

n3roGit, 2023-11-14:
| Switch 1 | Switch 2 | Function |
|:--------:|:--------:|:--------:|
| 2 Clicks | Hold     | Stepwise slower |
| Hold     | 2 Clicks | Stepwise faster |

Please adjust the code so that one side must always be held to change the speed. currently this is not necessary to hold.

happyherp, 2023-11-15:
See 7f8cd3af7657b445089ed14b603a24b0f25b2423 
With this change the speed change will only happen, if the other lever is also pressed **at the time that the other lever is released on the second click**. So you technically do not need to hold the other lever all the time, just at the very end. Implementing it in a way that the other button needs to be hold down the whole time takes more work.

n3roGit, 2023-11-15:
Perfect! Working!

n3roGit, 2023-11-15:
sorry for close

closed 2023-11-15 by n3roGit