BubTec/DPVControl #8: [FEATURE] More Click Codes

n3roGit, 2023-11-14:
Add Click Codes for:

| Switch 1 | Switch 2 | Function |
|:--------:|:--------:|:--------:|
| 1 Click  | 1 Click  |  cruise control       |
| 2 Clicks | 2 Clicks | Boost Mode |

Cruise control hold the actual speed till you press any button.
Boost mode jump to max speed but only how long you hold the button while pressing.
maybe its easier to implement "click click hold" to enable it insted of 3 click

n3roGit, 2023-11-17:
I've just tested cruise control. it's already working great! Please add that cruise control is also canceled if you press both buttons again. currently I have to press left or right.

happyherp, 2023-11-17:
> I've just tested cruise control. it's already working great! Please add that cruise control is also canceled if you press both buttons again. currently I have to press left or right.

I know. unfortunately that is very awkward to code. Technically, when you do anything, you leave cruise. So when you press down one lever, you already left. Then when you also click the other button at the same time, you enter cruise mode again.

happyherp, 2023-11-17:
Check a79ccb645fc679210c0aa98de0e25ad2f29ecb95 for turbo mode. 
When in normal mode, do click-click-hold on both buttons at the same time. it should enter turbo mode and accelerate to 100% duty. until you let go of any of the buttons. then it returns to its normal speed.

n3roGit, 2023-11-17:
almost perfect. the only thing missing is the setBarSpeed set to maximum during turbo. then we can merge

n3roGit, 2023-11-17:
and maybe add that the boost stays on as long as you keep pressing at least one button. but that's just a nice to have.

happyherp, 2023-11-17:
Both those things should work now. Try eab7424313f01328d96973a071e42a7cf1ae587f

closed 2023-11-17 by n3roGit