BubTec/DPVControl #9: [FEATURE] PreventOverload

n3roGit, 2023-11-14:
Finish PreventOverload function.

i need pre defined reductions for OverloadLimitMax like actual.

if motor draw more than OverloadLimit, the motor is spinning slower but if the load is lower it goes back to old value of speed.

happyherp, 2023-11-16:
What? Auf deutsch bitte.

n3roGit, 2023-11-16:
OverloadLimitMax = 40: definiert den Wert, den der Akku maximal liefern kann
OverloadLimit: Ist der Wert den der Motor benutzen kann ohne den Akku zu überlasten.

Da der VESC lediglich den Stromverbrauch des Motors messen kann und andere Verbraucher nicht einbezieht folgende Idee:

Es werden Bedingungen definiert, welche basierend auf statischen werden die maximal verfügbaren Ampere errechnen.

Bsp:
LED auf Stufe 3 verbraucht 3 A
OverloadLimit = OverloadLimitMax - LED auf Stufe 3
37                    = 40                           - 3

In diesem Beispiel darf der Motor maximal 37 Ampere aufnehmen.

Wird das Limit während des fahrens erreicht muss automatisch die Geschwindigkeit reduziert werden.

happyherp, 2023-11-19:
check 07b0548037f4985760301fb90ca0388e85310b99
I also added beeps at various points to alert the rider about what is going on.

n3roGit, 2023-11-19:
works perfectly. that i can put in production. currently there is still a merge problem

closed 2023-11-19 by happyherp