BubTec/DPVControl #15: [BUG] Motor acceleration

n3roGit, 2023-11-16:
const double DUTY_FACTOR = 1.0;
const double MIN_SPEED_DUTY = 0.38; //Duty on lowest setting.
const double MIN_DUTY_SOFT = 0.23; //Minumum Duty we sent to the motor during soft acceleration. 
const double DUTY_SOFT_SKIP = 0.15;

Please add the following here. I would like to be able to exclude a certain range in the DutyCycle, e.g. 0.15 - 0.23. Unfortunately, the engine stutters in this range. This allows me to guarantee a smooth start from 0.

In addition, the first speed level is currently not correct.
speed down: 1 
corresponds to 
currentMotorSpeed: 0.43 and not 0.38

happyherp, 2023-11-17:
try this: 592d2aaee92e4f046086101a1d33ef6d82efad86

n3roGit, 2023-11-17:
Hmm that still behaves totally stupid. I think we should go back to setRMP. that worked best =)

happyherp, 2023-11-21:
> In addition, the first speed level is currently not correct.
> speed down: 1
> corresponds to
> currentMotorSpeed: 0.43 and not 0.38

I fix this in production branch on a83f71a884d902662322ecad4a1c18942f48a526

happyherp, 2023-11-21:
See Commit 06fab3a49425af747aeffdc711c9a0dedc78a180 on the new branch. It uses RPM again.

n3roGit, 2023-11-22:
https://github.com/n3roGit/DPVControl/issues/37

n3roGit, 2023-12-30:
@happyherp  could we close this?

happyherp, 2023-12-30:
Yep. I will also delete the branch skip_duty_range where I did the changes.

closed 2023-12-30 by happyherp