BubTec/DPVControl #17: [FEATURE]  battery level

n3roGit, 2023-11-16:
`void updateBatteryLevel(float voltage) {
  if (voltage >= 5 && CellsInSeries >= 2) {
    float singleCellVoltages[] = {4.18, 4.1, 3.99, 3.85, 3.77, 3.58, 3.42, 3.33, 3.21, 3.00, 2.87};
    int singleCellPercentages[] = {100, 96, 82, 68, 58, 34, 20, 14, 8, 2, 0};

    for (int i = 0; i < sizeof(singleCellVoltages) / sizeof(singleCellVoltages[0]); i++) {
      if (voltage >= singleCellVoltages[i] * CellsInSeries) {
        // Interpolation
        if (i > 0) {
          float voltageRange = singleCellVoltages[i] * CellsInSeries - singleCellVoltages[i - 1] * CellsInSeries;
          int percentageRange = singleCellPercentages[i - 1] - singleCellPercentages[i];
          float voltageDifference = singleCellVoltages[i] * CellsInSeries - voltage;
          float interpolationFactor = voltageDifference / voltageRange;
          batteryLevel = singleCellPercentages[i] + interpolationFactor * percentageRange;
        } else {
          batteryLevel = singleCellPercentages[i];
        }
        break;
      }
    }
    // Ensure that the battery level is limited to the range [0, 100]
    batteryLevel = constrain(batteryLevel, 0, 100);
  } else {
    batteryLevel = 100;
  }
  int steps = (batteryLevel + 5) / LedBar2_Num;
  steps = constrain(steps, 0, LedBar2_Num - 1);
  setBarBattery(steps);
}`

I currently see a few improvements here. 

1. make singleCellVoltages and singleCellPercentages into one table. 
2. always use the average of the last X values to calculate the battery level. this is necessary because the voltage can drop briefly during acceleration. this means that for a moment, for example, only 30% of the battery is displayed, although 80% is actually still available. Here I would work with a measurement over at least the last 60 seconds.

n3roGit, 2023-11-17:
the output looks correct

n3roGit, 2023-11-17:
the calculation is as follows:

    float singleCellVoltages[] = {4.18, 4.1, 3.99, 3.85, 3.77, 3.58, 3.42, 3.33, 3.21, 3.00, 2.87};
    int singleCellPercentages[] = {100, 96, 82, 68, 58, 34, 20, 14, 8, 2, 0};
    
    
   The following is calculated for a 13S battery:
   
   40V / 13S = 3,08 -> between 8 and 2%

happyherp, 2023-11-18:
7227eb48cd6741c98750ec1d691655b5093dafec has a refactored soc calculation, which still works the same. It also records voltage and uses the average to calculate soc.

n3roGit, 2023-11-19:
implemented

closed 2023-11-19 by n3roGit