BubTec/DPVControl #18: [FEATURE] Write long time Data

n3roGit, 2023-11-16:
https://github.com/siara-cc/esp32_arduino_sqlite3_lib
https://medium.com/@ladolcefarniente/save-data-input-to-spiffs-with-esp32-and-arduino-ide-9e37011d7a91

Ideally, we can store a lot of data during operation on the 4MB memory of the ESP in order to analyze it after use. a possible variant could be to use the library linked above.

Spontaneously I can think of the following data for the database:
- Total operating time
- Motor Temp
- VESC Temp
- DHT22 Temp/Humidity
- UART.data.inpVoltage
- UART.data.avgInputCurrent
- UART.data.rpm
- Batterylevel

I

happyherp, 2023-11-19:
version on the branch starts a csv-file on startup. 
Every second we record time and temperature and write to it. 
Every startup uses a new file.
As far as I can tell there is no easy way to access the file system.

n3roGit, 2023-11-19:
`beepSequence:1
 still in standby: 1
Connected to VESC.
E (1027) SPIFFS: mount failed, -10025
writing to /datalog/data_0.csv
Listing directory: /datalog
  FILE: data_0.csv	SIZE: 0
Booting finished!
beepSequence:1
currentMotorSpeed: 0.00
Creating datapoint
Creating datapoint
Creating datapoint
Creating datapoint
Creating datapoint
Creating datapoint`

i got this after flashing

happyherp, 2023-11-19:
yep. that means its writing to a file. Now how do we read it? I would do it via wifi.

n3roGit, 2023-11-19:
Yes with wifi

happyherp, 2023-11-22:
I'll start the wifi implementation in this issue then. 

I like the [AdvancedWebsocket ](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/AdvancedWebServer/AdvancedWebServer.ino) better than working directly with the http-stream and having to write and parse http-headers myself like they do here: https://randomnerdtutorials.com/esp32-web-server-arduino-ide/

happyherp, 2023-11-22:
Having some trouble here .Stuck in this loop

```
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.print(WiFi.status());
  }

```
Output is status:1([WL_NO_SSID_AVAIL](https://www.arduino.cc/reference/en/libraries/wifi/wifi.status/)) and stays there. 

This guys seems to have a similiar problem: https://forum.arduino.cc/t/wifi-stopped-connecting-esp8266/593080/3
and fixed it by using another board firmware, I think. 

54758fa9469cf5dd64692999ea0452106abe8c1a this is the code i tried. @n3roGit can you check it out?

n3roGit, 2023-11-22:
This is because in this configuration the esp tries to dial into your wifi. However, we want to use the AP mode.

Have a look here:
https://github.com/n3roGit/DPVControl/issues/20

That's what I was referring to =)

happyherp, 2024-01-12:
We will need some kind of templating library to insert things into the html. There is the standard https://github.com/plapointe6/EspHtmlTemplateProcessor . But I don't think it will be enough. 

I would like to try https://github.com/me-no-dev/ESPAsyncWebServer . But that requires plattformIO, which is another vsCode plugin. So i will try that.

n3roGit, 2024-01-15:
I can try to do that. Can't we use the web server that is currently working?

happyherp, 2024-03-10:
[image]

I added the missing fields to the csv file. Motor values are all dummy, as I don't have one on my board. You should run it with the real thing, so we can adjust the formatting.  @n3roGit

n3roGit, 2024-03-10:
[image]
[logs (4).csv](https://github.com/n3roGit/DPVControl/files/14549693/logs.4.csv)

Looks great. i have niticed that i have two 0 csv files after delete. and i have attached the new generated log =)

happyherp, 2024-03-10:
the log looks ok, i would say. maybe we could display motor rpm without decimal places...

happyherp, 2024-03-10:
also added: display of file size in list; logging of led lamp state.

happyherp, 2024-03-10:
also logging: motorstate, speedsetting.

n3roGit, 2024-03-10:
Really Great!

Please add:

Battery Percentage
motorState
OverAllRuntime
WaterSensor State (Front and Back)

n3roGit, 2024-03-10:
[image]
[image]

Looks Nice. Why i have two files frome time to time?

happyherp, 2024-03-10:
> Looks Nice. Why i have two files frome time to time?

Not sure. But it is the file we are currently writing to. Can you live with that?

n3roGit, 2024-03-10:
> > Looks Nice. Why i have two files frome time to time?
> 
> Not sure. But it is the file we are currently writing to. Can you live with that?

Yes its OK

n3roGit, 2024-03-10:
> Really Great!
> 
> Please add:
> 
> Battery Percentage motorState OverAllRuntime WaterSensor State (Front and Back)

Only this would be usefull

happyherp, 2024-03-10:
> OverAllRuntime

That one is a bit more work. The timer resets with the machine. I would need to create a file to get the value from the last run. And regularly overwrite it to have the value for the next time, because we could be turned off at any moment. Please make a separate ticket for that one. 

> Battery Percentage
> WaterSensor State (Front and Back)

Those are easy.. will add.

n3roGit, 2024-03-10:
After the last update, the water sensor goes to 1 at the front and rear after starting.

n3roGit, 2024-03-10:
I have a simple solution to create graphs from the CSV:

https://webutility.io/csv-to-chart-online

Simply upload the file there as it is.

n3roGit, 2024-03-10:
@happyherp I have found another critical error in the current firmware. For some reason, both water sensors go TRUE. Even a restart does not fix this.

happyherp, 2024-03-10:
> I have found another critical error in the current firmware. For some reason, both water sensors go TRUE. Even a restart does not fix this.

Weird. I am pretty sure, that I did not change the code. But it looked weird to me as well, because it had SENSOR == LOW. 

I negated the sensors. Try again?

n3roGit, 2024-03-10:
> > I have found another critical error in the current firmware. For some reason, both water sensors go TRUE. Even a restart does not fix this.
> 
> Weird. I am pretty sure, that I did not change the code. But it looked weird to me as well, because it had SENSOR == LOW.
> 
> I negated the sensors. Try again?

No this isnt the solution. after the following commit its broken:

https://github.com/n3roGit/DPVControl/commit/0706cedec54c6233f299dc0e2ed16455dcd82113

n3roGit, 2024-03-10:
I suspect that it only occurs when the VESC is connected. It also looks like this in the log. 

If I leave the VESC connected while the ESP is already receiving power via USB, I can reproduce the error.

` Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Failed to get VESC data!: 0
 Loop took: 101
 Loop took: 116
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
 WARNING LEAK: 12121212
beepSequence:12121212
`