BubTec/DPVControl #19: [FEATURE] Emergency stop

n3roGit, 2023-11-16:
We need to monitor the value for getVescUart().data.rpm while the motor is running. If it is not approaching the desired speed as expected or if it suddenly stops, we assume that something has jammed. the motor must then be stopped immediately.

You could also use getVescUart().data.avgInputCurrent, as a strong fluctuation in the current consumption could also be an indication.

happyherp, 2023-11-21:
I put this in the motor_by_rpm branch. 
36f21eea86435f680d1b420bdd84f893a5a3935c

n3roGit, 2023-11-22:
https://github.com/n3roGit/DPVControl/issues/37

n3roGit, 2023-11-23:
> I put this in the motor_by_rpm branch. 
> 36f21eea86435f680d1b420bdd84f893a5a3935c
> 

Whats happening when motor is jammed on startup? In this case ampere rises and we have no RPM

n3roGit, 2024-01-06:
.

closed 2024-01-06 by n3roGit