BubTec/DPVControl #22: [BUG] Battery Level not calculated without USB Serial

n3roGit, 2023-11-18:
I have noticed that the LED indicator is not activated if I do not have a USB cable connected.

I have now narrowed it down to the fact that the updateBatteryLevel function is only executed when I have USB connected. 

I found this out by beeping when the function is executed.

`void updateBatteryLevel(float voltage) {
  if (voltage >= 5 && CellsInSeries >= 2) {
    float singleCellVoltages[] = {4.18, 4.1, 3.99, 3.85, 3.77, 3.58, 3.42, 3.33, 3.21, 3.00, 2.87};
    int singleCellPercentages[] = {100, 96, 82, 68, 58, 34, 20, 14, 8, 2, 0};

    for (int i = 0; i < sizeof(singleCellVoltages) / sizeof(singleCellVoltages[0]); i++) {
      if (voltage >= singleCellVoltages[i] * CellsInSeries) {
        // Interpolation
        if (i > 0) {
          float voltageRange = singleCellVoltages[i] * CellsInSeries - singleCellVoltages[i - 1] * CellsInSeries;
          int percentageRange = singleCellPercentages[i - 1] - singleCellPercentages[i];
          float voltageDifference = singleCellVoltages[i] * CellsInSeries - voltage;
          float interpolationFactor = voltageDifference / voltageRange;
          batteryLevel = singleCellPercentages[i] + interpolationFactor * percentageRange;
        } else {
          batteryLevel = singleCellPercentages[i];
        }
        break;
      }
    }
    // Ensure that the battery level is limited to the range [0, 100]
    batteryLevel = constrain(batteryLevel, 0, 100);
  } else {
    batteryLevel = 100;
  }
  int steps = (batteryLevel + 5) / LedBar2_Num;
  steps = constrain(steps, 0, LedBar2_Num - 1);
  setBarBattery(steps);
  beep("1111111111");  <------ HERE ----
}`

n3roGit, 2023-11-18:
if (UART.getVescValues()) {
    Serial.println("Connected to VESC.");
  } else {
    Serial.println("Failed to connect to VESC.");
    //hasMotor = false;
  }
  
  
  I was able to "solve" it. Apparently hasMotor is set to false at startup. Maybe we need to make a second function that also sets it back to true when connected.

happyherp, 2023-11-18:
That was something i build in to be able to work with my local setup without a motor. I don't quite understand, why it also is triggered if there is no usb.

happyherp, 2023-11-21:
I am not sure how to continue here.

n3roGit, 2023-11-22:
yes, that's strange. maybe we do that via a control file that is included in the gitignore. if "noMotor" file exists, then ignore it. This would also prevent the device from setting its dinest during operation in the event of communication problems at startup.

Or something with #ifdefine

n3roGit, 2023-11-23:
Or me make it in the gui to enable and disable it

n3roGit, 2023-12-30:
@happyherp Today I had something similar with motor jammed. The engine runs without the serial. Not with serial on.

happyherp, 2023-12-30:
I assume this is the same problem caused by getVescUart() returning a copy instead of a reference. I fixed that. Can you try it out?

3f0ad9f5f18cc0bd1925bff598ee3d7151f1cb4b

n3roGit, 2023-12-30:
bat lvl: 0
up 0 days, 0 hours, 1 minutes, 12 seconds
RPM: 0.00
inpVoltage: 0.00
ampHours: 0.00
tempMosfet: 0.00
tempMotor: 0.00
wattHours: 0.00
avgInputCurrent: 0.00
avgMotorCurrent: 0.00
dutyCycleNow: 0.00
Temp: 22.40°C
Humidity: 57.3%

Testen branch 22 but no differents

happyherp, 2024-01-02:
Is it still the same problem? Because I don't understand how you got a log-output without serial on. 
That is something we should fix anyways. Additionally to writing to serial, i should also be logging to the microcontrollers filesystem. Like we do in https://github.com/n3roGit/DPVControl/issues/18

closed 2024-01-06 by n3roGit