BubTec/DPVControl #26: Move away from arduino IDE

happyherp, 2023-11-19:
I dont like the IDE of Arduino that much. It was nice to get started, but i think there are better tools out there. 

I don't like the .ino files. It makes me create to many global variables. 
Also sometimes the arduino compiler seems to move declarations around that cause strange bugs.

happyherp, 2023-11-19:
I see there already is a branch https://github.com/n3roGit/DPVControl/tree/test-visual-studio-code

i am looking at this tutorial...
https://colddsam.medium.com/exploring-the-world-of-arduino-programming-in-visual-studio-code-bbcb4af982e7

happyherp, 2023-11-19:
What seems to work ok-ish.
VSCode with Arduido plugin from microsoft.
then install gcc compiler. https://nuwen.net/mingw.html
select the compiler in vscode.
then install the libraries that we use from inside vscode using the arduino library manager. If some are missing they need to be put in C:\Users\<user>\Documents\Arduino\libraries so that they can be found there. 

with that I can currently run verify and also upload to the board.
But there are MANY compilation errors. Most probably because of the .ino files. To fix this we would need to make them all .h/.cpp

happyherp, 2023-11-20:
i converted almost all code to cpp. 

Still my configuration shows wrong errors: 
[image]

n3roGit, 2023-11-21:
I think he is missing some default librarys. I search for this problem

n3roGit, 2023-11-21:
[image]
[image]

i got only this message with vscode branch

n3roGit, 2023-11-21:
check this https://github.com/n3roGit/DPVControl/commit/f5c7687e6f8adc7d1396db4811dcc097611e2076
[image]

with this its building bot not 100%
[image]

ö

n3roGit, 2023-11-21:
HA, Now its working!!!!!!!!

I can flash!

n3roGit, 2023-11-21:
Ah you need one thing:
https://downloads.arduino.cc/arduino-cli/arduino-cli_latest_Windows_64bit.msi
[image]
[image]
And set the CLI as compiler
C:\Program Files\Arduino CLI\

n3roGit, 2023-11-21:
After the changeover, communication with the vesc no longer works. we should take another look at this.

n3roGit, 2023-11-21:
tried:
`  if (getVescUart().getVescValues()) {
    Serial.println("Connected to VESC.");
  } else {
    Serial.println("Failed to connect to VESC.");
    hasMotor = true;
  }`
  
  
  and got this;
  
  
  ` Failed to get VESC data!: 0
 Failed to get VESC data!: 0`

n3roGit, 2023-11-21:
https://github.com/microsoft/vscode-arduino

i try it like this. working on it

n3roGit, 2023-11-21:
https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39

`- The supported legacy Arduino IDE versions are `1.6.x` and up to, but not including, `2.0.0`.
- The Windows Store's version of the Arduino IDE is not supported because of the sandbox environment that the application runs in.
- *Note:* Arduino IDE `1.8.7` had some breaking changes, causing board package and library installation failures.  These failures were corrected in `1.8.8` and later.
- *Note:* Arduino IDE `2.X.Y` is not supported and there are no plans for support in the future ([issue 1477](https://github.com/microsoft/vscode-arduino/issues/1477)).
`

happyherp, 2023-11-21:
[image]
That sums up the situation. Files are there, and vscode seems to know where. But CodeAnalysis does not find them.

happyherp, 2023-11-21:
> tried: ` if (getVescUart().getVescValues()) { Serial.println("Connected to VESC."); } else { Serial.println("Failed to connect to VESC."); hasMotor = true; }`
> 
> and got this;
> 
> ` Failed to get VESC data!: 0 Failed to get VESC data!: 0`

That is coming from somewhere else. 
See the other.cpp

```
void GetVESCValues() {

  if (!hasMotor) return;

  if (getVescUart().getVescValues()) {
    /*
    Serial.print("RPM: ");
    Serial.println(UART.data.rpm);
    Serial.print("inpVoltage: ");
    Serial.println(UART.data.inpVoltage);
    Serial.print("ampHours: ");
    Serial.println(UART.data.ampHours);
    Serial.print("tachometerAbs: ");
    Serial.println(UART.data.tachometerAbs);
    */

    updateBatteryLevel(getVescUart().data.inpVoltage);
  } else {
    log("Failed to get VESC data!", 00000);
  }
}
```

n3roGit, 2023-11-21:
> [image]
> That sums up the situation. Files are there, and vscode seems to know where. But CodeAnalysis does not find them. 

Maybe need to disable this feature?

happyherp, 2023-11-21:
I will disable it. I wanted to have the static c++ code analysis. But it is still better than Arduino IDE. I will merge what we have to prod and move on.

closed 2023-11-21 by happyherp